rust-embedded/meta-rust-bin #5: QA error: installed-vs-shipped

gautambt, 2016-10-09:
Hi,

I've added meta-rust-bin to by bblayers.conf and ran:

$ bitbake cargo-run

The command errored out with:

ERROR: cargo-bin-20160705-r0 do_package: QA Issue: cargo-bin: Files/directories were installed but not shipped in any package:
  /usr/lib
  /usr/etc
  /usr/lib/rustlib
  /usr/lib/rustlib/manifest-cargo
  /usr/share/zsh
  /usr/share/zsh/site-functions
  /usr/share/zsh/site-functions/_cargo
  /usr/etc/bash_completion.d
  /usr/etc/bash_completion.d/cargo
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
cargo-bin: 9 installed and not shipped files. [installed-vs-shipped]
ERROR: cargo-bin-20160705-r0 do_package: Fatal QA errors found, failing task.
ERROR: cargo-bin-20160705-r0 do_package: Function failed: do_package

Alsp cargo-bin seems to be installing files to /usr/etc instead of /etc:

build@126b11cf0ef1:~/build/tmp/work/cortexa9hf-neon-poky-linux-gnueabi/cargo-bin/20160705-r0/image$ find .
.
./usr
./usr/lib
./usr/lib/rustlib
....
./usr/etc
./usr/etc/bash_completion.d
./usr/etc/bash_completion.d/cargo
./usr/bin
./usr/bin/cargo

nastevens, 2016-10-10:
Hi @gautambt - thanks for the bug report.

Might I ask - what is the `cargo-run` target you're specifying? Do you mean `cargo-bin`?

It looks like you're trying to install cargo/rust on-target. (I'm basing that on the output directory of `cortexa9hf-neon-poky-linux-gnueabi`. Right now that is not tested/supported, and that is why you're getting packaging errors. I'd love to get that introduced, so PRs are welcome if you have time!

Have you had any success using `inherit cargo` to cross-compile packages for your target?

nastevens, 2016-10-10:
One other note - if you were just playing around and want to check the build, you can do `bitbake cargo-bin-native`. But the recommended way to use things right now is still `inherit cargo`

nastevens, 2018-01-18:
I think this issue has been fixed - please re-open if you are still seeing it.

closed 2018-01-18 by nastevens