rust-embedded/meta-rust-bin #8: Unable to build crates that depend on native libraries

kteza1, 2016-10-19:
I'm trying to build a crate which depends on systemd using below bitbake file

```
inherit cargo

SUMMARY = "GPIO Utilities"
HOMEPAGE = "git://github.com/rust-embedded/gpio-utils"
LICENSE = "CLOSED"

SRCREV="${AUTOREV}"
SRC_URI = "git://github.com/kteza1/systemd-test.git"
S = "${WORKDIR}/git"
```

Error:

```
= note: /usr/lib/gcc-cross/arm-linux-gnueabihf/5/../../../../arm-linux-gnueabihf/bin/ld: cannot find /lib/libpthread.so.0
| /usr/lib/gcc-cross/arm-linux-gnueabihf/5/../../../../arm-linux-gnueabihf/bin/ld: cannot find /usr/lib/libpthread_nonshared.a
| collect2: error: ld returned 1 exit status
```

I think system rootfs is being checked instead of built yocto rootfs. How do I fix this

posborne, 2016-10-19:
I am actually using gpio-utils within Yocto with meta-rust-bin currently and the following recipe works fine for me, although I am not using systemd:

If the code you are building needs to link against something like the systemd libraries, you will need to add an appropriate DEPENDS/RDPENDS to your recipe.  I don't see those.  In addition, make sure you are using the latest version of this layer as it includes a fix so pkg-config will be used to find proper paths.

kteza1, 2016-10-20:
Ok, this took some time to debug but mistake turned out to be a simple one. Instead of using `config` from `cargo_home` (set by meta-rust-bin), `config` from `~/.cargo/config` is being using which has wrong compiler set. 

I removed `~/.cargo` and everything worked. Is there a way to prevent this (make cargo not consider any other config except for `cargo_home/config`)?

nastevens, 2016-10-20:
The recipe exports the `CARGO_HOME` environment variable, which is supposed to override the default `~/.cargo`. The systems I've used for debugging have definitely had a `~/.cargo` directory and I haven't experienced this.

Could you attach the contents of the `log.do_compile` and `run.do_compile` from the `temp` directory of your recipe? You can find this in your build directory at `tmp/<arch>/<recipe>/<version>/temp`. Thanks!

nastevens, 2018-01-18:
@tekjar a recent change #26 may have improved this situation. Could you give it a try if you have some time?

nastevens, 2018-04-05:
Assuming that this has been fixed by #26  - if this isn't the case please feel free to reopen. Thanks!

closed 2018-04-05 by nastevens