rust-embedded/meta-rust-bin #10: nightly rust via meta-rust-bin

zzeroo, 2016-12-07:
Is it possible to get a nightly rust via meta-rust-bin? 

```
./build-new-version.sh nightly
```

fetches the checksums and build the files. But it looks yocto doesn't use these files. Sorry if this is the wrong place where I ask. And my yocto/ openembedded skills aren't  worth mentioning, too.

posborne, 2016-12-07:
Some additional work would be required to make nightly work, I think.  Shouldn't be too bad, but should be possible.  At a minimum, you would need to explicitly set your preferred version to select the nightly rust.

zzeroo, 2016-12-15:
What's the format for `build-new-version.sh` parameters? I mean
```
./build-new-version.sh nightly
```
works fine, with some little changes, but the recipe name looks odd. And _nightly_ is not a "explicitly preferred version" number.

posborne, 2016-12-16:
This is the correct form for calling the script, it's just that I think for nightly we would want to make some changes to allow for specifying which nightly, etc.

closed 2017-02-15 by zzeroo

sprhawk, 2021-12-09:
@zzeroo Please try this PR: https://github.com/rust-embedded/meta-rust-bin/pull/109