rust-embedded/meta-rust-bin #16: Wrong objcopy is run, not arm-poky-linux-gnueabi-objcopy

zzeroo, 2017-02-15:
A project of mine requires the [backtrace-rs](https://github.com/alexcrichton/backtrace-rs) crate. `backtrace-rs` itself need some C code compiled, and is using `objcopy` then. This call to `objcopy` fails, thats the error I try to report here. Maybe this is the wrong place.

This is the complete, failing, log: https://gist.github.com/zzeroo/c8fdc52f5c8d289b53c3eb1c3d778077
 I report the failure here: https://github.com/alexcrichton/backtrace-rs/issues/25#issuecomment-279742551 but I think its more related to the openembedded/yocto then the crate itself.

How can I force the usage of `arm-poky-linux-gnueabi-objcopy` instead of `objcopy`? As @alexcrichton mentioned here: https://github.com/alexcrichton/backtrace-rs/issues/25#issuecomment-279742551

nastevens, 2017-02-15:
Looking at the backtrace and code for `backtrace-rs` in [this function](https://github.com/alexcrichton/backtrace-rs/blob/master/backtrace-sys/build.rs#L17), it looks like the problem is the way that `backtrace-rs` is determining the name of the `objcopy` binary to use.  The name of the "binary" that `meta-rust-bin` uses is `cc-wrapper.sh`, a dynamically created shell script that contains the Yocto `$CC` command. `backtrace-rs` tries to replace the `-gcc` or `-cc` in that command name with `-objcopy`, resulting in something like `cc-objcopy`, which doesn't exist, so it falls back to just `objcopy`.

Quick background on _why_ we're using `cc-wrapper.sh` - Yocto outputs a "fat" `$CC` environment variable that has both the compiler (`arm-poky-linux-gnueabi-gcc` for example) with other machine and sysroot options like `-march=armv7-a`. The `gcc` crate expects `$CC` to be a single executable, not an executable with options, and silently drops the extra options specified in the `$CC` environment variable. Hence, we use the `cc-wrapper.sh` to encapsulate the Yocto `$CC` and provide a single executable to `gcc`. I personally don't like Yocto's decision to include options in `$CC` instead of `$CFLAGS`, but so many Yocto recipes now rely on the behavior that it would be a massive undertaking to change.

Some possible solutions:
- Update `backtrace-rs` to check for an `$OBJCOPY` environment variable and use that as the objcopy binary if it exists. Yocto outputs binary names for a lot of tools this way.
- Update `gcc` so that it accepts options after the binary in `$CC`. That means we could get rid of `cc-wrapper.sh`, but doesn't actually help with this specific problem since `backtrace-rs` is relying on `$CC` being a single executable name that it can parse into an `objcopy` binary name
- Expand functionality of `gcc` to include finding other common tools like `ar`, `objcopy`, etc. and then deal with cross-compiling concerns like this in that crate.

@alexcrichton - this affects a couple crates that you steward - do you have any thoughts?

alexcrichton, 2017-02-15:
I'd be totally fine with any of those solutions! I'm surprised that `gcc` is silently ignoring flags after the binary in `$CC`, that definitely sounds like a bug!

zzeroo, 2017-02-16:
@nastevens Thank you for the information. I've choosen the simplest way and changed `backtrace-rs`. But I hope I can fix the other (`gcc` and get rid of `cc-wrapper`), too. But this needs some more time. I miss the "big picture" for the meta-rust-bin parts and yocto/oe :/

closed 2017-02-16 by zzeroo