rust-embedded/meta-rust-bin #19: error: could not execute process `rustc -vV` (never executed)

zzeroo, 2017-05-16:
If I use meta-rust-bin git:master on the latest poky git:master the build fails with 
```
error: could not execute process `rustc -vV` (never executed)
```
The complete error log you found here: https://gist.github.com/e520b9932077678748a6eda91e27db79

I has bisect to this commit. 
http://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/?id=f15bcd78ba5f39bba80b539da43ac7ba0f7b3089

But I dont get whats happen. Here is the test recipe I use.

```
inherit cargo

SUMMARY = "Hello World in Rust"
HOMEPAGE = "https://github.com/zzeroo/hello-world-rs"
LICENSE = "GPLv2"
LIC_FILES_CHKSUM = "file://${COREBASE}/meta/COPYING.GPLv2;md5=751419260aa954499f7abaabaa882bbe"

SRC_URI = "git://git@github.com/zzeroo/hello-world-rs.git;protocol=ssh"

SRCREV = "12dae1d3e68511cc125893117dd44394d01c2927"
PR = "r4"
PV = "1.0+git${SRCPV}"

S = "${WORKDIR}/git"

# Fix: No GNU_HASH in the elf binary
INSANE_SKIP_${PN} = "ldflags"
INSANE_SKIP_${PN}-dev = "ldflags"
```

posborne, 2017-05-29:
I have seen errors like this before when a tool try to execute the combination of the command (`rustc`) and its args (`-vV`) together as a command with spaces that does not exist.  I'll need to dig in a bit more to figure out what might be going on.

With python, this is sometimes seen when a string is used for a command like "rustc -vV" without `shell=True` set.  The bisected commit doesn't seem related, but 🤷‍♂️

nastevens, 2018-01-18:
Hey @zzeroo - would you be able to try with the code from #27? I was experiencing the same errors on some code using Yocto Pyro and I think that branch has the needed changes.

nastevens, 2018-02-26:
#27 is merged, so hopefully these issues have been addressed. Please reopen if that's not the case. Thanks!

closed 2018-02-26 by nastevens