rust-embedded/meta-rust-bin #21: Building Rust library that is contained within existing CMake project?

robertsonb, 2017-09-16:
Hello -

Im working on an existing C/CMake project that has a Rust library within it. Currently the CMake project builds the Rust component by driving Cargo (doesn't use rustc). So to build it one just have to run the usual 'make' and off it goes.

I have to get this running on an embedded target via Yocto, and Im trying to avoid splitting the project into hundreds of small recipes (one for each lib). So I just made a single recipe, and used the included CMake builder with the intent of just running the build as normal.

Of course there is the Rust component, for which I figured I would just use the meta-rust-bin layer. So I inherit both the CMake builder and the Rust/Cargo builder into the single recipe and build.

The problem (as you might have guessed) is that the CMake builder runs and the cross compiling environment for Rust doesn't seem to be setup properly; so all artifacts are for the host system, not the target (guessing it stems from inheriting both 'cmake' and 'cargo' in a single recipe).

The only option I have thought of at this point is to just make a recipe for the Rust component, build that first, and then kick off the CMake build and link against the built Rust library.

But before I do that I wanted to see if there is a way to use the meta-rust-bin layer to solve this problem a little cleaner/elegantly. Thoughts?

robertsonb, 2017-09-17:
Ended up just using values inherited from cargo.bbclass to feed CMake triple/sysroot/search-paths, and all is well.

closed 2017-09-17 by robertsonb

nastevens, 2017-09-17:
Thanks for bringing this up @robertsonb. I know that @posborne has done some work with CMake and Rust. In that case, I believe that the hierarchy is reversed, i.e. CMake kicks off Cargo to do the Rust build as a static library, and then CMake handles the final linking.

It sounds like you got to a working solution so 👍