rust-embedded/meta-rust-bin #48: rustc and cargo not found?

cwaldren, 2019-04-12:
This is an extremely newbie question, so pardon that. 

When I try to bitbake my recipe, I get errors that `cargo` and `rustc` can't be found. This makes sense to me, because I don't have them installed on my build machine, so they certainly wouldn't magically be in `$PATH`.

I was under the impression that the precompiled binaries that are fetched by `cargo-bin-cross-` and `rust-bin-cross-` would be used for compilation. I know I'm having a pretty basic misunderstanding here.

Thanks!

nastevens, 2019-04-13:
If your code contains a Cargo.toml and is buildable by Cargo, you shouldn't even need to call directly to `cargo` or `rustc` - take a look at the example here: https://github.com/rust-embedded/meta-rust-bin#basic-example. The "magic" is coming from [`cargo.bbclass`](https://github.com/rust-embedded/meta-rust-bin/blob/master/classes/cargo.bbclass), which is brought into the recipe with `inherit cargo`.

Once you have `inherit cargo`, you should also get `cargo` and `rustc` on your `$PATH` for the recipe you've added the `inherit` into. If you need to do something more complicated, I recommend reading the code for `cargo.bbclass` and seeing how it's doing things or ask follow up questions here.

Hoping this is enough to get you started! Documentation PRs are also always welcome - when you've worked with this stuff long enough it's easy to forget what someone starting out might find difficult.

cwaldren, 2019-04-13:
To clarify, I followed the basic example pretty much word for word, and my Rust project is just the default hello world generated by cargo init.

When I run `bitbake my_recipe`, that's when I'm getting the 'cargo: command not found`.

I didn't explicitly call cargo or rustc from the command line - it appeared to be failing on the invocations in `cargo_do_compile`.

I think somehow the path isn't getting set correctly, perhaps through my build script (which calls oe-init-build-env followed by some recipe cleans and then my recipe.)

nastevens, 2019-04-13:
Hmm... that is interesting and answers my next question which was to see if you'd tried a "hello world" example. What you've described should work totally fine.

Which version of Yocto are you using? It's possible there's been a non-compatibility with the latest versions potentially. Also, if you're able to share your project and point me to it, I'm more than willing to clone and try building myself.

cwaldren, 2019-04-14:
Unfortunately I can't share the project. It's using Jethro, so we're definitely behind the times :) 

I'll keep looking into it. If I figure it out I'll post here or submit a PR.

nastevens, 2019-04-14:
You should still be okay on Jethro - even Fido would work, and there's a (outdated) branch for Dizzy even.

On my side I'll try to find some time to fire up a hello world project to test with. If nothing else it would serve as a good example.

cwaldren, 2019-04-23:
I did some more experimenting today, with devshell.

I inspected the environment variables, and there was no mention of cargo or rust's binary paths. I did, however, see the `CARGO_HOME`, `RUST_BACKTRACE`, `CARGO_TARGET_DIR` variables that are set up in the `cargo.bbclass`.

What's the actual mechanism that is supposed to put `rustc` and `cargo` into the path?

I can almost see the light!

posborne, 2019-04-23:
@cwaldren The PATH stuff should be setup as a result of the DEPENDS on `cargo-bin-cross-<ARCH>`.  Might be worth checking that you aren't overrding the DEPENDS_append from the bbclass in your own recipes.

cwaldren, 2019-04-25:
It seems like the DEPENDS is working, as when I build my recipe I always see `cargo-bin-cross-arm` being fetched and compiled, and the "Rust is ready to roll" message. 

Then when my recipe is built, I see:
```
++ which rustc

+ bbnote 'which rustc:'

+ '[' -p /apps_proc/poky/build/tmp-glibc/work/armv7a-vfp-neon-oe-linux-gnueabi/rusthello/1.0-r0/temp/fifo.105003 ']'
+ printf '%b\0' 'bbnote which rustc:'

NOTE: which rustc:
++ rustc --version
/apps_proc/poky/build/tmp-glibc/work/armv7a-vfp-neon-oe-linux-gnueabi/rusthello/1.0-r0/temp/run.do_compile.105003: line 124: rustc: command not found
```

I've attached the full verbose log below (~200 lines, `cargo-bin-cross-arm` + `rusthello` recipe.) 

I would really appreciate if one of you could take a quick look and see if anything comes to mind!

[bitbake.log](https://github.com/rust-embedded/meta-rust-bin/files/3118327/bitbake.log)

1kloc, 2020-01-11:
@cwaldren could you give these (https://github.com/1kloc/rust-yocto-examples.git) examples a try?  I had a similar issue, the solution for me was primarily setting `protocol=git;` in the SRC_URI. 

- Slightly modified version of [basic example](https://github.com/1kloc/rust-yocto-examples/blob/master/yocto/meta-rust-examples/recipes-examples/gpio-utils/gpio-utils_0.3.0.bb)
- Another example of a [simple hello cargo recipe](https://github.com/1kloc/rust-yocto-examples/blob/master/yocto/meta-rust-examples/recipes-examples/hello-cargo/hello-cargo_git.bb)

1kloc, 2020-01-14:
@cwaldren
https://github.com/rust-embedded/meta-rust-bin/commit/8e44c2e8e3a9f27c4661a4414a9d5f7d0959ceab
 appears necessary for jethro, are you using master or dizzy branch in meta-rust-bin ?
I was able to get meta-rust-bin working with jethro by after merging dizzy changes into master

cwaldren, 2020-01-14:
@1kloc , switching to the `dizzy` branch immediately solved the issue on Jethro.

closed 2020-01-14 by cwaldren