rust-embedded/meta-rust-bin #75: cross compiling using bindgen under x86_64, but requires 32 bit

sprhawk, 2020-06-21:
I'm building [btrfsutil-rs](https://github.com/sprhawk/btrfsutil-rs), 
it complains following errors:
```
| --- stdout
| cargo:rustc-link-lib=btrfsutil
| cargo:rerun-if-changed=btrfsutil_wrapper.h
| cargo:rerun-if-changed=bindgen_whitelist.toml
|
| --- stderr
| /usr/include/x86_64-linux-gnu/gnu/stubs.h:7:11: fatal error: 'gnu/stubs-32.h' file not found
| /usr/include/x86_64-linux-gnu/gnu/stubs.h:7:11: fatal error: 'gnu/stubs-32.h' file not found, err: true
| thread 'main' panicked at 'Unable to generate bindings: ()', deps/btrfsutil-rs/build.rs:44:39
| stack backtrace:

```

I can build ffi for [libimobiledevice](https://github.com/libimobiledevice/libimobiledevice) without problem

sprhawk, 2020-06-21:
I have used `BINDGEN_EXTRA_CLANG_ARGS` to fixed it.

Is this a bug or intention ?

nastevens, 2020-07-16:
Can you share a minimal example that demonstrates this please?

sprhawk, 2020-08-28:
I was busy doing project. I'll try to make a small example project recently

hellow554, 2020-09-16:
Same issue here 🙋 

In my case, I have a program, that depends on [socketcan-alt](https://github.com/Hakuyume/socketcan-rs), which uses bindgen, to produce bindings for the `<can.h>` header.

Here's the error message for me (which is somewhat similar to @sprhawk s)

```
| error: failed to run custom build command for `socketcan-alt v0.2.2 (https://github.com/Hakuyume/socketcan-rs.git?rev=f79cef958e7f4ec68ee1e0f30a7b66df15411ad6#f79cef95)`
| 
| Caused by:
|   process didn't exit successfully: `/home/op/y_unic/build/tmp/work/cortexa7t2hf-neon-vfpv4-mut-linux-gnueabi/emv-test/1.0-r0/target/release/build/socketcan-alt-31d2e4f812a3f006/build-script-build` (exit code: 101)
|   --- stderr
|   /usr/include/linux/types.h:5:10: fatal error: 'asm/types.h' file not found
|   /usr/include/linux/types.h:5:10: fatal error: 'asm/types.h' file not found, err: true
```

I'm building for an arm device, so like sprhawk not for the same platform. Maybe that's the issue?

A question, which you might be able to answer: What target is bindgen aiming at? The host or target platform, when he's looking for the `<linux/can.h>` file (e.g. does he try to use the `recipe-sysroot` or `recipse-sysroot-native` directory?)

hellow554, 2020-09-16:
It seems to me, that he uses my systems `/usr/include/linux/types.h` file, instead of the `sysroot` ones. Is that possible? I don't think, that that is inteded, right?

ryankurte, 2020-09-17:
this appears to be a common issue with bindgen, may be related to the "Pre-built bindgen outputs often do not match target architecture sizes" section [here](https://github.com/rust-embedded/wg/issues/481).

sprhawk, 2021-01-31:
use following defines in layer to fixed it:
```
export BINDGEN_EXTRA_CLANG_ARGS
BINDGEN_EXTRA_CLANG_ARGS = "--sysroot=${WORKDIR}/recipe-sysroot -I${WORKDIR}/recipe-sysroot/usr/include"
```

closed 2021-01-31 by sprhawk

hellow554, 2021-01-31:
Hey @sprhawk 
That looks like a proper fix for this Problem. IMHO this should be included into this layer.
What do you think @nastevens ?

nastevens, 2021-02-10:
@hellow554 I'm not sure that it belongs directly in this layer, but I am more than happy to accept a PR to add something to the README since I suspect that there is a lot of crossover between people using Yocto and needing to interface Rust with C.

If this were to be included in the layer I'd rather see it as a `.bbclass` specifically for supporting bindgen. I don't want to lose track of this idea so I've opened #96.