rust-embedded/meta-rust-bin #81: Make use of Profile Name Overrides in Config/Env

posborne, 2020-08-13:
https://doc.rust-lang.org/stable/cargo/reference/config.html#profilenamebuild-override is now available so some of the things we would like to do in this layer are now possible on stable.  Examples include forcing builds to be done with debug symbols so we can do the proper split debug later, etc.

CC @nastevens

nastevens, 2020-08-14:
Related to #64

DavidAntliff, 2024-02-28:
> Examples include forcing builds to be done with debug symbols 

Can you give an example of how to do this with `meta-rust-bin` please?

nastevens, 2024-02-28:
@DavidAntliff This is actually an enhancement to allow this functionality with `meta-rust-bin`. I've added the correct label now.

Or were you looking for guidance on how to add this feature?

DavidAntliff, 2024-02-28:
@nastevens I am looking for guidance, yes, please. Ultimately I want to enable debug builds via bitbake/devtool: https://github.com/rust-embedded/meta-rust-bin/issues/167