rust-embedded/meta-rust-bin #86: Clean up old releases

otavio, 2020-08-27:
I am wondering if it makes sense we support so many old releases (Rust versions)?

I am not against providing some old releases but we ought to have a limit and also make sure those _work_. What is your take on this?

otavio, 2020-08-28:
I prepared following commit, in my tree:

```
commit db8250943cb2b58edef1f1c62418d49ae9782ac8 (HEAD -> fix/issue-70)
Author: Otavio Salvador <otavio@ossystems.com.br>
Date:   Thu Aug 27 21:17:30 2020 -0300

    Clean up old Rust releases
    
    We are removing Releases up to Rust 1.39.0. We also removed releases
    which we had minor bug fix releases.
    
    In summary, the following versions were deleted:
    
     - 1.11.0
     - 1.12.0
     - 1.12.1
     - 1.14.0
     - 1.15.0
     - 1.15.1
     - 1.16.0
     - 1.21.0
     - 1.22.1
     - 1.23.0
     - 1.24.0
     - 1.25.0
     - 1.26.2
     - 1.27.2
     - 1.28.0
     - 1.30.0
     - 1.30.1
     - 1.31.0
     - 1.32.0
     - 1.33.0
     - 1.34.0
     - 1.34.2
     - 1.35.0
     - 1.36.0
     - 1.37.0
     - 1.38.0
     - 1.39.0
     - 1.41.0
     - 1.45.0
     - 1.45.1
    
    The following releases as kept:
    
     - 1.40.0
     - 1.41.1
     - 1.44.0
     - 1.45.2
     - 1.46.0
    
    Signed-off-by: Otavio Salvador <otavio@ossystems.com.br>
```

Does it makes sense?

ryankurte, 2020-08-29:
@posborne not super familiar with this component but, seems reasonable to me to drop versions < our `embedded-hal` MSRV?

otavio, 2020-08-29:
What is the `embedded-hal` MSRV currently?

ryankurte, 2020-08-29:
uhh, it [appears](https://github.com/rust-embedded/embedded-hal/blob/d81cf7cdb36d2214136891c7a6e0d45f9fe04723/.github/workflows/ci.yml#L21) to be `1.35.0`

hberntsen, 2020-08-30:
There is no real cost in keeping old versions, right? I only see a bunch of small files which is nothing in the scope what a standard Yocto build generates.

It would be nice to have some automated tests to guarantee that a given version still works. At that point it would make more sense to me to list the old versions as untested. When someone notices they don't work anymore they can be fixed or dropped. As a reference, `rustup` is also stil supporting a lot (all?) of old versions.

otavio, 2020-08-31:
It is not just about cost but usefulness. If nobody uses it, it makes no sense to be kept there.  If someone has a real need for an old version, it is just a matter of running the script. Now the question is, why I'd wish to keep using a so old release?

hberntsen, 2020-08-31:
From that perspective: I don't know who will need older versions. Maybe some people will want match the version their distro uses (see https://repology.org/project/rust/versions ). As far as I know the Rust developers only support the latest version (i.e., there is no Rust LTS version).

otavio, 2020-08-31:
Sure and the main question here is if we'll handle bug reports on old releases. I can say I won't use the time to fix an issue with Rust 1.30 for example. 

Leving it there just for being there, seems pointless.

hberntsen, 2020-09-02:
True, but then I'd argue the project should be open to others willing to maintain old releases. 

It seems we're not getting any other feedback on this topic. Maybe apply the commit you suggested and see if anyone complains?

posborne, 2020-09-02:
I'm fine with dropping the older releases; if someone is using an older version they are most likely not updating the layer anyway as everything on the older version is working fine with the system they are building.  At present, things in the layer are simple enough that we generally haven't had problems, but CI is also not in place to confirm that (which would be nice to have but a bit of an undertaking to setup)

@nastevens You agree?

nastevens, 2020-09-02:
I think looking at the Yocto project itself lends to the idea that we don't need to maintain every version forever since releases regularly remove old versions of recipes. I think things are a little bit different with Rust since new features are added so frequently and embedded developers are likely to want to pick a version and stick with it for longer than other developers.

So I think what I'm saying is that going with "most recent minus 2 versions" seems excessive to me, but something like pruning releases older than 2 years or something like that I don't have a problem with.

otavio, 2020-09-03:
I ended removing releases up to 1.39.0; let me know if I should rework it somehow. Personally, I don't see the reason to keep so many old releases as users can use specific layer revisions if desired. I think we ought to boost the use of newer toolchain as it ensures we keep users up to date.

As I said before, leaving the recipes there without ensuring they work seems pointless and as we are using binaries we cannot fix issues found if not related to metadata and packaging and Rust itself does not keep maintaining old toolchains so I'd prefer to stick to it as well.

PR #88 (2020-09-04): Clean up old Rust releases
We are removing Releases up to Rust 1.39.0. We also removed releases
which we had minor bug fix releases.

In summary, the following versions were deleted:

 - 1.11.0
 - 1.12.0
 - 1.12.1
 - 1.14.0
 - 1.15.0
 - 1.15.1
 - 1.16.0
 - 1.21.0
 - 1.22.1
 - 1.23.0
 - 1.24.0
 - 1.25.0
 - 1.26.2
 - 1.27.2
 - 1.28.0
 - 1.30.0
 - 1.30.1
 - 1.31.0
 - 1.32.0
 - 1.33.0
 - 1.34.0
 - 1.34.2
 - 1.35.0
 - 1.36.0
 - 1.37.0
 - 1.38.0
 - 1.39.0
 - 1.41.0
 - 1.45.0
 - 1.45.1

The following releases as kept:

 - 1.40.0
 - 1.41.1
 - 1.44.0
 - 1.45.2
 - 1.46.0

Fixes: #86
commit 864cfe6f (2020-09-04): Clean up old Rust releases

We are removing Releases up to Rust 1.39.0. We also removed releases
which we had minor bug fix releases.

In summary, the following versions were deleted:

 - 1.11.0
 - 1.12.0
 - 1.12.1
 - 1.14.0
 - 1.15.0
 - 1.15.1
 - 1.16.0
 - 1.21.0
 - 1.22.1
 - 1.23.0
 - 1.24.0
 - 1.25.0
 - 1.26.2
 - 1.27.2
 - 1.28.0
 - 1.30.0
 - 1.30.1
 - 1.31.0
 - 1.32.0
 - 1.33.0
 - 1.34.0
 - 1.34.2
 - 1.35.0
 - 1.36.0
 - 1.37.0
 - 1.38.0
 - 1.39.0
 - 1.41.0
 - 1.45.0
 - 1.45.1

The following releases as kept:

 - 1.40.0
 - 1.41.1
 - 1.44.0
 - 1.45.2
 - 1.46.0

Fixes: #86
closed 2020-09-04 by posborne