rust-embedded/meta-rust-bin #96: Incorporate bindgen support

nastevens, 2021-02-10:
Adapted from #75

Presumably there is a large crossover between people using meta-rust-bin and people who need to interface C with Rust. Providing a "paved path" for using bindgen would therefore be useful.

I am unaware of any existing recipes/layers to support bindgen in yocto right now but would be happy to be proven wrong.

hellow554, 2021-02-10:
https://github.com/rust-embedded/meta-rust-bin/issues/75#issuecomment-776399157
> I'm not sure that it belongs directly in this layer

I mean, where else? This layer is about compiling software with Rust. That includes bindgen as well, because it's part of the rust ecosystem.
So IMHO it should be `meta-rust-bin`s job to tell clang/bindgen where the real sysroot is.

I can see, that this is very specific to *one* package (namely bindgen) and other may have different quirks (I don't know anything about cbindgen or similar).

So, either we can have a generic environment variable, e.g. `CLANG_ARGS` which will be accepted by all packages that need to deal with it, or we try to reach the maintainer of said packages and ask for their 2 cents on this topic.

hellow554, 2021-08-03:
ping @nastevens :)

TopperBG, 2021-10-12:
Any workout here?
I'm investigate simillar issue with crosscompilation of redisjson and pretty sure that one could be the root of bindgen problems
https://github.com/rust-lang/rust-bindgen/pull/1915
But it's implented patch already ?