rust-embedded/meta-rust-bin #97: Support for build scripts

allsey87, 2021-03-08:
Is it the case at the moment, that any package using a [build script](https://doc.rust-lang.org/cargo/reference/build-scripts.html) will not compile?

With an empty project, it seems that as soon as I add a dependency on a package that includes a `build.rs`, the build fails. For example, if I add [log](https://github.com/rust-lang/log) as a dependency, I get the output:
```
 DEBUG: Executing shell function do_compile
 NOTE: which rustc: /home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/recipe-sysroot-native/usr/bin/rustc
 NOTE: rustc --version rustc 1.49.0 (e1884a8e3 2020-12-29)
 NOTE: which cargo: /home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/recipe-sysroot-native/usr/bin/cargo
 NOTE: cargo --version cargo 1.49.0 (d00d64df9 2020-12-05)
 NOTE: cargo build --verbose --manifest-path /home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/Cargo.toml --target=x86_64-unknown-linux-gnu --release
     Updating crates.io index
  Downloading crates ...
   Downloaded log v0.4.14
   Downloaded cfg-if v1.0.0
    Compiling log v0.4.14
    Compiling cfg-if v1.0.0
      Running `rustc --crate-name build_script_build /home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/log-0.4.14/build.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 -C debug-assertions=off -C metadata=72f27afcef2156d3 -C extra-filename=-72f27afcef2156d3 --out-dir /home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/release/build/log-72f27afcef2156d3 -C linker=/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/wrappers/linker-wrapper.sh -L dependency=/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/release/deps --cap-lints allow`
      Running `rustc --crate-name cfg_if --edition=2018 /home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/cfg-if-1.0.0/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no -C debuginfo=2 -C metadata=bb222c302d676c2e -C extra-filename=-bb222c302d676c2e --out-dir /home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/x86_64-unknown-linux-gnu/release/deps --target x86_64-unknown-linux-gnu -C linker=/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/wrappers/linker-wrapper.sh -L dependency=/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/x86_64-unknown-linux-gnu/release/deps -L dependency=/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/release/deps --cap-lints allow`
      Running `/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/release/build/log-72f27afcef2156d3/build-script-build`
 error: failed to run custom build command for `log v0.4.14`
 
 Caused by:
   could not execute process `/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/release/build/log-72f27afcef2156d3/build-script-build` (never executed)
 
 Caused by:
   No such file or directory (os error 2)
 WARNING: exit code 101 from a shell command.
 ERROR: Execution of '/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/temp/run.do_compile.61272' failed with exit code 101:
     Updating crates.io index
  Downloading crates ...
   Downloaded log v0.4.14
   Downloaded cfg-if v1.0.0
    Compiling log v0.4.14
    Compiling cfg-if v1.0.0
      Running `rustc --crate-name build_script_build /home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/log-0.4.14/build.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 -C debug-assertions=off -C metadata=72f27afcef2156d3 -C extra-filename=-72f27afcef2156d3 --out-dir /home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/release/build/log-72f27afcef2156d3 -C linker=/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/wrappers/linker-wrapper.sh -L dependency=/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/release/deps --cap-lints allow`
      Running `rustc --crate-name cfg_if --edition=2018 /home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/cfg-if-1.0.0/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no -C debuginfo=2 -C metadata=bb222c302d676c2e -C extra-filename=-bb222c302d676c2e --out-dir /home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/x86_64-unknown-linux-gnu/release/deps --target x86_64-unknown-linux-gnu -C linker=/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/wrappers/linker-wrapper.sh -L dependency=/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/x86_64-unknown-linux-gnu/release/deps -L dependency=/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/release/deps --cap-lints allow`
      Running `/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/release/build/log-72f27afcef2156d3/build-script-build`
 error: failed to run custom build command for `log v0.4.14`
 
 Caused by:
   could not execute process `/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/release/build/log-72f27afcef2156d3/build-script-build` (never executed)
 
 Caused by:
   No such file or directory (os error 2)
 WARNING: exit code 101 from a shell command.
```

allsey87, 2021-03-08:
What I find a bit strange about this error, however, is that it claims that `/home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/release/build/log-72f27afcef2156d3/build-script-build` could not be found. However it does exist and is executable:
```
ls -al /home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/release/build/log-72f27afcef2156d3/build-script-build
-rwxr-xr-x 2 mallwright mallwright 3269416 Mar  8 15:24 /home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/release/build/log-72f27afcef2156d3/build-script-build
```

nastevens, 2021-03-08:
This sounds similar to https://github.com/rust-embedded/meta-rust-bin/issues/94 - could you please double-check that you've updated to the latest `master`?

allsey87, 2021-03-08:
Hi @nastevens, I am indeed on the latest master, you can confirm that from the yocto build configuration: 
```
meta-rust-bin = "master:f8e79ed922c765fca224b2318ab172e0b59f4a4d"
```
Perhaps this is an issue with build host and target architecture being (almost) the same?

allsey87, 2021-03-09:
This is the output of `ldd`. While it seems like this issue is related to #94 it doesn't appear to be related to #56 
```
ldd /home/mallwright/poky/build/tmp/work/corei7-64-poky-linux/hello-rust/1.0.0-r0/target/release/build/log-72f27afcef2156d3/build-script-build
	linux-vdso.so.1 (0x00007fff20be0000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fa2a1fa4000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fa2a1d85000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fa2a1b81000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fa2a1790000)
	/lib/ld-linux-x86-64.so.2 => /lib64/ld-linux-x86-64.so.2 (0x00007fa2a21bc000)
```
I suspect that #94 is a duplicate of this issue but was solved by some configuration changes unrelated to #56 and was incorrectly closed.

allsey87, 2021-03-09:
Digging in a bit deeper, I can confirm that `RUST_TARGET` and `RUST_BUILD` are both set to `x86_64-unknown-linux-gnu`. I am able to get the build to complete successfully by changing `linker-wrapper.sh` to  `linker-wrapper-native.sh`, that is:
```diff
diff --git a/classes/cargo.bbclass b/classes/cargo.bbclass
index bcb2c05..9f62759 100644
--- a/classes/cargo.bbclass
+++ b/classes/cargo.bbclass
@@ -61,7 +61,7 @@ create_cargo_config() {
     else
         echo > ${CARGO_HOME}/config
         echo "[target.${RUST_TARGET}]" >> ${CARGO_HOME}/config
-        echo "linker = '${WRAPPER_DIR}/linker-wrapper.sh'" >> ${CARGO_HOME}/config
+        echo "linker = '${WRAPPER_DIR}/linker-native-wrapper.sh'" >> ${CARGO_HOME}/config
     fi
 
     echo >> ${CARGO_HOME}/config
```
Although I am skeptical as to whether the final binary will run on my target. I will test this now.

allsey87, 2021-03-09:
Indeed, this doesn't work and reports file not found, which as I understand, means that the loader was not found.
```
root@up-core:~# which hello-rust
/usr/bin/hello-rust
root@up-core:~# ls -al `which hello-rust`
-rwxr-xr-x    1 root     root        280888 Mar  9 10:34 /usr/bin/hello-rust
root@up-core:~# hello-rust
-sh: /usr/bin/hello-rust: No such file or directory
```

allsey87, 2021-03-09:
@nastevens as you mentioned in your [comment](https://github.com/rust-embedded/meta-rust-bin/issues/56#issuecomment-521004482), the change in 2c460c05f4d0a8563fc9269e15c94728e2422074 causes problems for build.rs in dependency crates.

otavio, 2021-03-25:
@allsey87 last time I looked at this, we got non-build-script crates working. It seems we are blocked on Rust itself not allowing to have individual target setups for the same target.

allsey87, 2021-03-26:
@otavio do you know if there been an issue opened for this on the Rust's/cargo's repo?

otavio, 2021-03-26:
@allsey87 yes, I found it in the past but didn't find it now. Sorry.

rich-g, 2021-05-20:
@allsey87 @otavio Seems like this might be the issue in question? https://github.com/rust-lang/cargo/issues/3349.

otavio, 2021-05-20:
@rich-g exactly. It has a patch under work to fix it.

rich-g, 2021-06-01:
Looks like the right tooling is now in place in nightly: https://github.com/rust-lang/cargo/blob/master/src/doc/src/reference/unstable.md#host-config

otavio, 2021-06-01:
Yes, however, I believe we cannot use it stable until it is release. Am I wrong?

nastevens, 2021-06-16:
> Yes, however, I believe we cannot use it stable until it is release. Am I wrong?

You're correct for how things work at present. You can work around it as described in https://github.com/rust-embedded/meta-rust-bin/issues/101 but I wouldn't feel comfortable adding the changes more generally until this is stabilized. That said I'm definitely open to workarounds using current stable if anyone has any ideas.

sabra-james, 2021-08-18:
The fix is available as of rust `1.54.0` stable, you can enable it like [this](https://github.com/buildroot/buildroot/commit/4b2be770b8a853a7dd97b5788d837f0d84923fa1).

jameshilliard, 2021-08-18:
The fix is available as of rust `1.54.0` stable, you can enable it like [this](https://github.com/buildroot/buildroot/commit/4b2be770b8a853a7dd97b5788d837f0d84923fa1).

posborne, 2023-10-10:
Should be fixed with the merge of #146

closed 2023-10-10 by posborne