rust-embedded/meta-rust-bin #101: Support nightly version of rust

cybertanyellow, 2021-04-27:
Is it possible to support nightly version of rust? the main purpose is to build [rocket web framework](https://github.com/SergioBenitez/Rocket/issues/19)
Or if have supported, how to configure in meta-rust-bin?

Thanks,

nastevens, 2021-05-03:
We don't currently support nightly, and there's a couple of reasons for that:
* Right now generation of new releases is manual and it's unlikely we'd ever be able to keep up with daily updates. So automation of new releases would have to come first.
* The versioning is messy because of how channel information is provided by rust-lang. We request `https://static.rust-lang.org/dist/channel-rust-<version>.toml` to get the manifest of files we need to download. You can absolutely request `https://static.rust-lang.org/dist/channel-rust-nightly.toml`, but unlike fixed release versions, that's going to change every day. That makes recreating past nightly releases difficult if not impossible. And in Yocto not being able to recreate a build is usually a bad thing, so not something we'd considered before.
* Us old embedded developers are scared of anything that hasn't been released for at least a decade or so (mostly kidding)

That said, you are not out of luck! The `build-new-version.sh` script in the root of the repo will work just fine with nightly. So you can do
```sh
./build-new-version.sh nightly
```
and create your own recipes for nightly toolchains. If you're only targeting one architecture you can remove unneeded ones from https://github.com/rust-embedded/meta-rust-bin/blob/master/build-new-version.sh#L22

Let me know how it goes!

closed 2021-06-16 by nastevens

sprhawk, 2021-12-09:
@cybertanyellow try this PR: https://github.com/rust-embedded/meta-rust-bin/pull/109