rust-embedded/meta-rust-bin #114: Typo in cargo.bbclass?

TheComet, 2021-12-20:
```sh
DEPENDS:append = "\
    cargo-bin-cross-${TARGET_ARCH} \
    pkgconfig-native \
"
```

Bitbake fails to parse ```DEPENDS:append```, did you mean ```DEPENDS_append```?

TheComet, 2021-12-20:
Ok, I didn't realize bitbake introduced a new override syntax.

Unfortunately, OpenSTLinux has not migrated to it yet so we are forced to use a version of bitbake that does not enforce it yet. The easiest solution for me is to use the iss114 branch I created that reverts this line to the old syntax.

closed 2021-12-20 by TheComet

nastevens, 2022-02-18:
Sorry, I seem to have missed this PR when you originally opened it. We've got a `zeus` branch that is before the the change to the new syntax and I'm planning to backport changes for at least a bit longer. Maybe that'd work for you.