rust-embedded/meta-rust-bin #134: Unable to compile a native recipe

krsh, 2022-11-22:
When I compile a native application, I get this error:

```
ERROR: Nothing PROVIDES 'cargo-bin-native-x86_64' (but virtual:native:/home/test/meta-example/recipes-app/app/app.bb DEPENDS on or otherwise requires it)
ERROR: Required build target 'core-image-minimal' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-minimal', 'app-native', 'cargo-bin-native-x86_64']
```

Other info:

```
Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "qemuarm64"
DISTRO               = "poky"
DISTRO_VERSION       = "4.0.3"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa57"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       
meta-oe              
meta-rust-bin        = "<unknown>:<unknown>"
```

samliddicott, 2022-12-01:
I find that a -native recipe can be compiled simply by adding:

PROVIDES += "cargo-bin-native-${TARGET_ARCH}"

to recipes-devtools/rust/cargo-bin-cross.inc

I'm not sure it is the right fix, but it is a simple one.

mangodan2003, 2024-05-16:
I have a pull request open for a change we've been using for a while, it works for both x86 and arm (Raspberry pi) builds. 
https://github.com/rust-embedded/meta-rust-bin/pull/178

nastevens, 2024-05-21:
Resolved by #178. Thanks @mangodan2003!

closed 2024-05-21 by nastevens