rust-embedded/meta-rust-bin #135: Not working on langdale as bbclasses with same name inherited/layered incorrectly

dellgreen, 2022-11-23:
Same problem as #108 

Having the exact same problem on poky 'langdale' with meta-rust-bin master branch (after testing with LAYERSERIES_COMPAT updated). Even with the fixes, the layering is always putting poky layer first on my BBPATH irrespective of how things are ordered in my bblayers.conf and the prepend change in layer.conf, and any BBMASK-ING. I can get it to work if I rename the cargo.bbclass and rust-common.bbclass to something unique, so they don't clash with poky rust bbclasses (as recommended in the bitbake dev manual)

dellgreen, 2022-11-23:
I can get meta-rust-bin to now be first on the BBPATH but now this  breaks the meta rust bbclasses when being evaluated

dellgreen, 2022-11-23:
I stand corrected, i cannot get it to build

banditopazzo, 2022-11-28:
I think that the meta-rust-bin should have completely different class names and variables names to avoid collision with meta-rust. 

If you uses some other rust application recipe depending on the official meta-rust, it's not possible to mask that layer.

You should be allowed to use both meta-rust and meta-rust-bin in your image, leaving the single application depending on the one it needs

DavidAntliff, 2023-09-23:
@dellgreen did you find a way forward with this? I need to find a way to use rust 1.70 or newer in Langdale.

dellgreen, 2023-09-23:
> @dellgreen did you find a way forward with this? I need to find a way to use rust 1.70 or newer in Langdale.

No I removed rust from the project I was working on as it was only a single rust executable I wanted to compile, and found other way to replicate it's functionality

PR #136 (2023-10-07): Classes rename
# Summary

In this PR the class names were changed to unique names to avoid collision with the official `meta-rust` in the modern yocto releases. The goal is to be able to use both `meta-rust` and `meta-rust-bin` in the same project.

## Added
- `langdale` in the `LAYERSERIES_COMPAT` variable

## Changed  
- rename `rust-common` class to `rust_bin-common`
- rename `cargo` class to `cargo_bin`

Fix #135
closed 2023-10-07 by posborne