rust-embedded/meta-rust-bin #145: `mips-unknown-linux-gnu` is unavailable for download for channel `stable`

jdswensen, 2023-09-12:
I haven't dug into this too much, but it looks like `mips-unknown-linux-gnu` might not be a valid target anymore. Or maybe there's a misconfiguration upstream. But I get a `404` when running the update script for `1.72.0`. Additionally, when I manually added the toolchain to my system (with an older version installed) and performed an update:

```
info: syncing channel updates for 'stable-x86_64-unknown-linux-gnu'
info: latest update on 2023-08-24, rust version 1.72.0 (5680fa18f 2023-08-23)
error: component 'rust-std' for target 'mips-unknown-linux-gnu' is unavailable for download for channel 'stable'
error: If you don't need the component, you could try a minimal installation with:
error:
error:     rustup toolchain add stable --profile minimal
```

nastevens, 2023-09-25:
I know there was talk about dropping MIPS to Tier 2, and it appears that has happened: https://doc.rust-lang.org/nightly/rustc/platform-support.html. Since the whole idea of meta-rust-bin is to use upstream-compiled stable binaries we can probably just drop MIPS support from 1.72.0 onward. LMK if you'd like to take this work or not, just so there's no duplication of effort.

jdswensen, 2023-10-05:
Hey @nastevens! I figured that _might_ be the case, but didn't know where to look. 😄 I can tackle this tonight if you haven't already. Should just require removing the target triple `mips-unknown-linux-gnu`, right?

jdswensen, 2023-10-06:
Looks like `mipsel-unknown-linux-gnu` also 404'd.

jdswensen, 2023-10-06:
PR here: https://github.com/rust-embedded/meta-rust-bin/pull/147

commit 59f03315 (2023-10-06): fix: build: remove unsupported binary triples

The following triples are no longer upstream-compiled:

- `mips-unknown-linux-gnu`
- `mipsel-unknown-linux-gnu`

They have been removed from the `TARGET_TRIPLES`.

Fixes: #145
closed 2023-10-06 by posborne