rust-embedded/meta-rust-bin #151: Cross compilation broken since #146

igiona, 2023-10-11:
Since #146 our build is broken.

```shell
| error: linker `cc` not found
|   |
|   = note: No such file or directory (os error 2)
| 
```

Reverting to 7ac87ecf6c05252b5070c9f6733cff01819f998a the build works again.

Our setup
- Host: Ubuntu 22.0
Running on: https://github.com/actions/runner-images/blob/main/images/linux/Ubuntu2204-Readme.md#rust-tools
- Target armv7-unknown-linux-gnueabihf

It looks like the linker configuration is not properly set.
In the changes you switched from the "environment variable configuration" approach to the config file approach.
From the rust documentation, it seems that the config file should be called `config.toml` : https://doc.rust-lang.org/cargo/guide/cargo-home.html , here it seems to be called `config`.
Could this be the issue pherhaps?

igiona, 2023-10-11:
@Benji-NW I came to your same conclusion and it fixed in the same way.
I also created a PR for this: https://github.com/rust-embedded/meta-rust-bin/pull/152

eldruin, 2023-12-11:
Closing since this was fixed in #152.
Thank you and feel free to comment or reopen if there are additional concerns.

closed 2023-12-11 by eldruin