rust-embedded/meta-rust-bin #159: QA Error of strip image

HosseinAssaran, 2023-12-27:
When packaging the recipe for[ bat project ](https://github.com/sharkdp/bat)it gives the below error 
`ERROR: bat-0.24.0-r0 do_package: QA Issue: File '/usr/bin/bat' from bat was already stripped, this will prevent future debugging! [already-stripped]
ERROR: bat-0.24.0-r0 do_package: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: [/home](tg://bot_command?command=home)[/hossein](tg://bot_command?command=hossein)[/test_yocto_project](tg://bot_command?command=test_yocto_project)[/build](tg://bot_command?command=build)-qemu-aarch64/tmp/work/cortexa57-poky-linux/bat/0.24.0-r0/temp/[log.do](https://log.do/)_package.98360
ERROR: Task (/home/hossein/test_yocto_project/meta-custom/recipes-app/bat/[bat_0.24.0.bb](https://bat_0.24.0.bb/):do_package) failed with exit code '1'`
I think it is because it makes release but yocto need debug binary

nastevens, 2023-12-28:
Hi @HosseinAssaran - could you please share your recipe file? `meta-rust-bin` does build in release mode but it also [enables debug symbols by default](https://github.com/rust-embedded/meta-rust-bin/blob/8a6a14481a8cf2cceafaec0049087ef797ba191f/classes/cargo_bin.bbclass#L107) and should work without QA issues. Is it possible you overwrote a variable instead of extending it by accident? (i.e. used `=` instead of `+=` or similar)

HosseinAssaran, 2023-12-30:
Sure. This is the content of my recipe file named as bat_0.24.0.bb which generate QA error
```
# Summary from project homepage
SUMMARY = "A cat clone with wings"

# for the sake of package manager info we set SECTION
SECTION = "tools"

HOMEPAGE = "https://github.com/sharkdp/bat"

# According to the README fie of the project
LICENSE = "MIT"

# Enable network for the compile task allowing cargo to download dependencies
do_compile[network] = "1"

SRC_URI = "git://github.com/sharkdp/bat.git;protocol=https;branch=master"
SRCREV = "28990bc4512770a2669de33cf71a2edb26abd061"

LIC_FILES_CHKSUM = "file://LICENSE-MIT;md5=c46eaa1315aaa0c727a29b157ad9170a"

do_install() {
	install -d ${D}${bindir}
	install -m 0755 bat ${D}${bindir}
}

inherit cargo_bin
```

HosseinAssaran, 2023-12-30:
I found that this behaviour is becasue of  `strip = true` is added to release profile. You can see it in the below refrence:

https://github.com/sharkdp/bat/blob/bf56cd90f06e8450280265fd53e4fafb08bad536/Cargo.toml#L111

nastevens, 2023-12-30:
> I found that this behaviour is becasue of strip = true is added to release profile of bat project

Interesting... Yeah, that would definitely cause the behavior that you're seeing. Some quick background on why this is a problem when building under Yocto. Yocto expects binaries to have debug symbols and then it does its own strip step but saves the stripped debug symbols. That way you are able to debug binaries from your released system without the storage space cost of shipping the debug symbols as part of your OS, so called "detached" debug symbols. 

The other side of the problem is the `strip = true` line in bat, which causes Cargo to run the strip step itself and discard the debug symbols. This happens before the Yocto strip step which is where the error comes in. I don't think bat is wrong for doing this - it makes sense for their builds to strip the release binaries before shipping. I think it's just a bad interaction where neither side is doing the "wrong" thing but they still interact poorly.

If you want the detached debug symbols to work I think the easiest way is going to be creating and applying a patch to the bat project that removes the `strip = true` line from the `Cargo.toml`. You can save this as a `.patch` file and add it to your `SRC_URI` so Yocto will automatically apply it during the build. The other option if you're okay without debug symbols is to add `INSANE_SKIP_${PN} += "already-stripped"` to your recipe, which should silence the QA error.

Just kicking around the idea, but it might not be a bad idea to have meta-rust-bin check for this setting so that the produced error is more helpful. But unless there's something I'm missing I don't see a good way to have meta-rust-bin just automatically "do the right thing". Any thoughts @rust-embedded/embedded-linux?

HosseinAssaran, 2023-12-31:
Thank you, @nastevens, for your comprehensive and helpful response. I've successfully resolved the issue by modifying the [EXTRA_RUSTFLAGS](https://github.com/rust-embedded/meta-rust-bin/blob/8a6a14481a8cf2cceafaec0049087ef797ba191f/classes/cargo_bin.bbclass#L24) introduced in the `cargo_bin.bbclass`. In my recipe, I set `EXTRA_RUSTFLAGS` to "-C strip=none" to bypass the stripping process, eliminating the need to make changes to either of the projects involved. So there is no need for the pullrequest #160.

Here's the line added to the bat project recipe:

```bb
EXTRA_RUSTFLAGS = "-C strip=none"
```
You can find the final version of my recipe [here](https://github.com/HosseinAssaran/meta-custom-yoctotraining/blob/main/recipes-app/bat/bat_0.24.0.bb).

Again, thank you for your assistance!

closed 2023-12-31 by HosseinAssaran