rust-embedded/meta-rust-bin #171: Can't build rust project without modiying rust_bin-common.bbclass to remove the check for os == 'linux'

Egod3, 2024-03-22:
I am experiencing an odd issue using the meta-rust-bin layer with bitbake/Yocto (Kirkstone branch) and trying to build a simple Rust Application.  We are building Linux with Yocto for the BeaglePlay (Aarch64) and the build machine is x86_64. 

When building the image I get this error:

```
ERROR: /workspace/yocto/nqma/fli-nqma-yocto-build/build_beagleplay/../layers/meta-rust-bin/recipes-devtools/rust/rust-bin-cross_1.73.0.bb: Unsupported OS: eabi. Only Linux is supported.                                                                                                                                      
ERROR: Parsing halted due to errors, see error messages above
```

I dug into the file that is printing that error message and found this line:

```
if not os == 'linux':
      bb.fatal("Unsupported OS: %s. Only Linux is supported." % os)
```

If I add a line above it to print what os is, it shows up "Linux" but the line is checking for "linux" lowercase.  

I can comment out the line and get my build to work but that doesn't feel the right solution.  I ran my rust application and it is working too, although it is pretty basic right now. 

Any suggestions for things to check to fix this build error?

nastevens, 2024-03-22:
Thanks for the bug report! Could you please try again with the [changes from this branch I just pushed](https://github.com/rust-embedded/meta-rust-bin/compare/bugfix/ignore-os-case)? There are some unintentionally case-sensitive checks that I've fixed, I think that's what you're encountering here.

nastevens, 2024-03-23:
@Egod3 The fix is merged to master now, so please try that. If that didn't fix your error feel free to reopen this issue. Thanks again!

closed 2024-03-23 by nastevens