rust-embedded/meta-rust-bin #184: How to select version

frosch01, 2024-07-23:
Hi,

when I check the code, it seems there are many different cargo version supported at the same time. Can you please update documentation on how a version can be selected. I was not able to find an answer looking at the Yocto documentation. 

Thx

Ralph

seimonw, 2024-07-23:
I use yocto's PREFERRED_VERSION variable, set in my local.conf:

`PREFERRED_VERSION_rust-bin-cross-arm = "1.76.0"`
`PREFERRED_VERSION_cargo-bin-cross-arm = "1.76.0"`

robagar, 2024-09-04:
Is it possible to select the rust version on a per-recipe basis?  Generally I want to use latest stable rust, but some things need a fixed version (specifically in our case, Zenoh needing to built with rust 1.75)

seimonw, 2024-09-04:
I don't think there's an easy way to do this, to my knowledge you can't have two versions of a package at the same time.
Only thing I could think of is something like creating your own cargo-bin-175 class and pulling that into your zenoh recipe instead of the regular class.
Feels a bit messy, but might work!

nastevens, 2024-09-04:
There's no guide for it at the moment, but it _should_ be possible to use Yocto multiconfig to do this. There's a short guide [here](https://prashant-divate.medium.com/yocto-multiconfig-platform-development-building-for-versatility-134b1dbb2062) and more details in the Yocto documentation itself.

If you get this working it'd be awesome to see your notes on how!

seimonw, 2024-09-05:
Yeah, that works @nastevens 

I was able to create a `build/conf/multiconf/rust-175.conf`:
```
PREFERRED_VERSION_rust-bin-cross-arm = "1.75.0"
PREFERRED_VERSION_cargo-bin-cross-arm = "1.75.0"
```
Add `BBMULTICONFIG = "rust-175"` to my local.conf

Then `bitbake mc:rust-175:myrustdemo`

nastevens, 2024-09-09:
That's great, thanks for testing that out @seimonw!

MirkoDonnarumma, 2025-03-18:
Hello guys, thank you for your effort.
@seimonw I tried what you did but ended up with an error.

### my local.conf:

``` conf
BBMULTICONFIG = "rust-175"
```

### my conf/multiconfig/rust-175.conf:

``` conf
PREFERRED_VERSION_rust-bin-cross-arm = "1.75.0"
PREFERRED_VERSION_cargo-bin-cross-arm = "1.75.0"
```

Then I ran `bitbake mc:rust-175:myrustapp` and I got this:

``` log
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /home/qtmmidonnar/Progetti/qm-reborn/poky/build/tmp/work/cortexa53-fslc-linux/quantum-manager/0.1.0.AUTOINC+c9064c40bd-r0/recipe-sysroot-native/usr/bin/cargo
| NOTE: rustc =
| NOTE: cargo build -v --target aarch64-fslc-linux --release --manifest-path=/home/qtmmidonnar/Progetti/qm-reborn/poky/build/tmp/work/cortexa53-fslc-linux/quantum-manager/0.1.0.AUTOINC+c9064c40bd-r0/git//Cargo.toml
| error: package `libc v0.2.171` cannot be built because it requires rustc 1.63 or newer, while the currently active rustc version is 1.59.0
| WARNING: exit code 101 from a shell command.
```

Did it happen to you as well? If so, how did you resolve this?

The branches I am using are all "kirkstone" while the latest commit I am using for meta-rust-bin is master:85a33ed0725c8e88f8e103c347d207c602f8b86e

I hope you can help me because i've been trying to make this work for the past 3 weeks (yeah, i'm a newbie lol)

Thank you for your time