rust-embedded/meta-rust-bin #194: Target 'thumbv6m-none-eabi' not supported

Knaifhogg, 2025-01-07:
Applies to more than thumbv6m-none-eabi but it's the one needed for a STM32 microcontroller.

From quickly looking at it the 'rust_target' function in rust_bin-common would need to be able to return 'thumbv6m-none-eabi' so the cargo build call gets the correct target. From modifying things locally it seems changing TARGET_ARCH and TARGET_OS to match causes PROVIDES issues. So it looks like more bitbake knowledge is needed to implement a fix.

nastevens, 2025-01-07:
@Knaifhogg Could you expand on what you're trying to do? `meta-rust-bin` is a project to support Rust compilation for Yocto Linux, so trying to support a `none` OS target doesn't really make sense.

Knaifhogg, 2025-01-08:
My build is mainly aarch64 linux, however I have one recipe of a C project that cross-compiles to thumbv6m. This is done by inheriting allarch and providing correct ARM toolchain for the makefile in the project. This binary is then stored in the linux filesystem so it can flash units on its own. I am looking to replace the C project with one in Rust so everything is built at the same time.

I am new to bitbake so my approach is to swap the recipe so the Rust project builds a binary just as the project it replaces. I don't know if the original way of building is a good way :smile:

nastevens, 2025-01-10:
Oh I see, you need to have the toolchain available to do the bare-metal build. I don't currently see a good way to do this with the meta-rust-bin APIs.

It seems like a reasonable feature request though: a way to specify extra Rust targets to install beyond just those available for the Yocto OS version.