rust-embedded/meta-rust-bin #204: Regression: Can no longer build libsqlite3-sys

taamrooz, 2025-02-27:
Hey there,

After updating the layer to the `feature/rust-1.85.0` branch coming from 1.82.0, I noticed I couldn't build one of my projects' dependency crate `libsqlite3-sys`.

The full output is below:

```sh
error: failed to run custom build command for `libsqlite3-sys v0.30.1`
|
| Caused by:
|   process didn't exit successfully: `/home/user/r/yocto-repo/build/tmp-glibc/work/armv8a-linux/my-rust-project/0.1-r0/target/release/build/libsqlite3-sys-6b39159f4007bc55/build-script-build` (exit status: 1)
|   --- stdout
|   cargo:rerun-if-env-changed=LIBSQLITE3_SYS_USE_PKG_CONFIG
|   cargo:include=/home/user/r/yocto-repo/build/tmp-glibc/work/armv8a-linux/my-rust-project/0.1-r0/cargo_home/registry/src/index.crates.io-1949cf8c6b5b557f/libsqlite3-sys-0.30.1/sqlite3
|   cargo:rerun-if-changed=sqlite3/sqlite3.c
|   cargo:rerun-if-changed=sqlite3/wasm32-wasi-vfs.c
|   cargo:rerun-if-env-changed=SQLITE_MAX_VARIABLE_NUMBER
|   cargo:rerun-if-env-changed=SQLITE_MAX_EXPR_DEPTH
|   cargo:rerun-if-env-changed=SQLITE_MAX_COLUMN
|   cargo:rerun-if-env-changed=LIBSQLITE3_FLAGS
|   OUT_DIR = Some(/home/user/r/yocto-repo/build/tmp-glibc/work/armv8a-linux/my-rust-project/0.1-r0/target/release/build/libsqlite3-sys-5f597ba6f07b69c2/out)
|   TARGET = Some(x86_64-unknown-linux-gnu)
|   OPT_LEVEL = Some(0)
|   HOST = Some(x86_64-unknown-linux-gnu)
|   cargo:rerun-if-env-changed=CC_x86_64-unknown-linux-gnu
|   CC_x86_64-unknown-linux-gnu = None
|   cargo:rerun-if-env-changed=CC_x86_64_unknown_linux_gnu
|   CC_x86_64_unknown_linux_gnu = None
|   cargo:rerun-if-env-changed=HOST_CC
|   HOST_CC = None
|   cargo:rerun-if-env-changed=CC
|   CC = Some(/home/user/r/yocto-repo/build/tmp-glibc/work/armv8a-linux/my-rust-project/0.1-r0/wrappers/cc-wrapper.sh)
|   RUSTC_WRAPPER = None
|   cargo:rerun-if-env-changed=CC_ENABLE_DEBUG_OUTPUT
|   cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
|   CRATE_CC_NO_DEFAULTS = None
|   DEBUG = Some(true)
|   CARGO_CFG_TARGET_FEATURE = Some(fxsr,sse,sse2)
|   cargo:rerun-if-env-changed=CFLAGS_x86_64-unknown-linux-gnu
|   CFLAGS_x86_64-unknown-linux-gnu = None
|   cargo:rerun-if-env-changed=CFLAGS_x86_64_unknown_linux_gnu
|   CFLAGS_x86_64_unknown_linux_gnu = None
|   cargo:rerun-if-env-changed=HOST_CFLAGS
|   HOST_CFLAGS = Some()
|   cargo:rerun-if-env-changed=CC_SHELL_ESCAPED_FLAGS
|   CC_SHELL_ESCAPED_FLAGS = None
|   cargo:warning=aarch64-unknown-linux-gcc: error: unrecognized command-line option '-m64'
|
|   --- stderr
|
|
|   error occurred: Command "/home/user/r/yocto-repo/build/tmp-glibc/work/armv8a-linux/my-rust-project/0.1-r0/wrappers/cc-wrapper.sh" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-gdwarf-4" "-fno-omit-frame-pointer" "-m64" "-DSQLITE_CORE" "-DSQLITE_DEFAULT_FOREIGN_KEYS=1" "-DSQLITE_ENABLE_API_ARMOR" "-DSQLITE_ENABLE_COLUMN_METADATA" "-DSQLITE_ENABLE_DBSTAT_VTAB" "-DSQLITE_ENABLE_FTS3" "-DSQLITE_ENABLE_FTS3_PARENTHESIS" "-DSQLITE_ENABLE_FTS5" "-DSQLITE_ENABLE_JSON1" "-DSQLITE_ENABLE_LOAD_EXTENSION=1" "-DSQLITE_ENABLE_MEMORY_MANAGEMENT" "-DSQLITE_ENABLE_RTREE" "-DSQLITE_ENABLE_STAT4" "-DSQLITE_SOUNDEX" "-DSQLITE_THREADSAFE=1" "-DSQLITE_USE_URI" "-DHAVE_USLEEP=1" "-DHAVE_ISNAN" "-D_POSIX_THREAD_SAFE_FUNCTIONS" "-DHAVE_LOCALTIME_R" "-DSQLITE_ENABLE_UNLOCK_NOTIFY" "-o" "/home/user/r/yocto-repo/build/tmp-glibc/work/armv8a-linux/my-rust-project/0.1-r0/target/release/build/libsqlite3-sys-5f597ba6f07b69c2/out/0343851c9008f270-sqlite3.o" "-c" "sqlite3/sqlite3.c" with args cc-wrapper.sh did not execute successfully (status code exit status: 1)
```

My target is `aarch64-unknown-linux-gnu` but for some reason `TARGET` is set to `Some(x86_64-unknown-linux-gnu)`, which is my host system.
The `-m64` option should only be enabled for `x86-64`, `aix` or `powerpc64`, see: https://github.com/rust-lang/cc-rs/blob/8ea5e95c10d0a6b169233736e681be5442af9671/src/lib.rs#L2149

When looking at the commits between 1.82.0 and 1.85.0 I saw this commit: https://github.com/rust-embedded/meta-rust-bin/commit/96c802da6a582978721e60b96487d460c8eadf38. After reverting this commit I was able to build successfully on 1.85.0.

To reproduce:
- Create Rust project with sqlx crate and the sqlite feature.
- Create a recipe which inherits cargo_bin from the current `master` branch.
- Build for `aarch64-unknown-linux` with a `x86-64-unknown-linux` host system.

imdanielsp, 2025-04-01:
Hi all, we are having the same issue as well. Just +1 the issue.

cesar-pa, 2025-04-07:
Same issue in our code base. +1.